PickNikRobotics/rviz_visual_tools #32: Issues with axes display functions

Nandite, 2016-06-16:
When we use publishAxis() functions to display a large number of items (100 for example), only a few part of the axes will be display. It can randomely be one third or a half of the axes which are going to appear on RViz. For example i just reuse your rviz_visual_tools_demo.cpp code like that : 
```
ROS_INFO_STREAM_NAMED(name_, "Displaying Coordinate Axis");
    pose1.translation().x() = 0;
    y += space_between_rows;
    pose1.translation().y() = y;
    step = 0.025;
    for (double i = 0; i <= 10.0; i += step)
    {
      visual_tools_->publishAxis(pose1);
      if (!i)
        publishLabelHelper(pose1, "Coordinate Axis");

      pose1.translation().x() += step;
      pose1 = pose1 * Eigen::AngleAxisd(step * 2 * M_PI, Eigen::Vector3d::UnitX()) *
              Eigen::AngleAxisd(step * 2 * M_PI, Eigen::Vector3d::UnitY()) *
              Eigen::AngleAxisd(step * 2 * M_PI, Eigen::Vector3d::UnitZ());
    }
    visual_tools_->triggerBatchPublish();

```

By putting `i <= 10.0` i should normally have 400 axes displayed on the screen, but we only have few of them randomly drawn on the screen on the x axis.

davetcoleman, 2016-06-16:
it could be the ROS message buffer is getting overrun  - try moving the triggerBatchPublish() inside the for loop and adding a short sleep timer just in case

Nandite, 2016-06-16:
@VictorLamoine , Done it with a timer of 100 ms. It does not change the behavior. It seems that the issues is coming specifically from axes. Indeed i've made the test for all other forms and items, and they work perfectly.

davetcoleman, 2016-06-29:
Any resolution to this issue?

VictorLamoine, 2016-06-30:
No, we are working on a demonstrator that is due for the 11th of July, we'll get back to this after!

davetcoleman, 2016-09-28:
has this issue been resolved?

VictorLamoine, 2016-09-29:
Yes, this issue was fixed, not sure by which commit (I didn't closely look at your latest modifications)
Test code [here](https://github.com/VictorLamoine/rviz_visual_tools/commit/c2f1fafddde2bd7520c541df97bc6abec16e72f).

e98d3e81b0290b9ad82bfac841d30a17c1d63f42: (replace `trigger` by `triggerBatchPublish` in the test code)
[image]
[image]
[image]

c57d92a88e07b49d4f6d90310e1d840fe3381517 (latest) + my test code commit (c2f1fafddde2bd7520c541df97bc6abec16e72ff):
[image]

Please test and close if you can reproduce both the issue and the fix.

davetcoleman, 2016-09-29:
This is probably just dropped ROS messages - if you blast Rviz with too many separate markers you will sometimes lose some, depending on your buffer size. My recent work has switched from publishing markers every time you create a maker, to only when you call "trigger()", to allow batching of markers much easier.

closed 2016-09-29 by davetcoleman

VictorLamoine, 2017-06-01:
Memo for my future self: stop reducing `rviz_visual_tools` markers queue size to 1 in RViz, messages will be dropped and incomplete markers will be displayed.