PickNikRobotics/rviz_visual_tools #33: Delete marker by Id

Nandite, 2016-06-16:
Currently when we call the function deleteAllMarkers(), it remove every markers added in RViz. It would be great to have a function deleteMarkerById(), since we do not want to remove marker drawn on the screen at one time.

Sincerely

davetcoleman, 2016-06-16:
Yes that would be a good feature to add - Rviz would support it!

VictorLamoine, 2016-06-16:
It would also be nice to have a DeleteAllBut and specifying a list of ID's to keep

davetcoleman, 2016-06-16:
This would be more complicated because we'd have to track all object IDs that have been published and maintain that list throughout - I'm not sure the overhead is worth it. In my workflow, I always just delete everything and redraw the scene

davetcoleman, 2016-07-05:
I do not plan on implementing this, are you guys working on an implementation?

VictorLamoine, 2016-07-05:
No not for now, maybe in the future

closed 2016-07-05 by davetcoleman