PickNikRobotics/rviz_visual_tools #35: Merging rviz_visual_tools and moveit_visual_tools into one unified repo

davetcoleman, 2016-06-16:
moveit_visual_tools uses the same parent class RvizVisualTools so would be a good candidate to merge together. I'm trying to reduce the time I spend maintaining and releasing lots of separate repos. The downside to this merge is that by default you will have to bring in lots of moveit dependencies when you do a git clone. The workaround for this group of users will be either to:

- simply add CATKIN_IGNORE to moveit_visual_tools
- delete the folder
- use catkin-tools' blacklist feature

Thoughts?

Note: one of my main motivations for doing this is for practice for a much bigger [moveit merge](http://discourse.ros.org/t/migration-to-one-github-repo-for-moveit/266/11) that I've been working on. I've already made the script that will merge the git histories together.

VictorLamoine, 2016-06-29:
:+1: as long as we are still able to use it without MoveIt

Other suggestion:
- adding a CMake option "WITH_MOVEIT_VISUAL_TOOLS"

shaun-edwards, 2016-10-20:
Did you ever merge this repo with MoveIt or is it still independent?

davetcoleman, 2016-10-20:
still independent. i was going to do it as "practice" for the moveit merge, but that's long done and i think i lost interest in merging these repos...

closed 2016-10-20 by davetcoleman