PickNikRobotics/rviz_visual_tools #36: JacobiSVD error

VictorLamoine, 2016-06-28:
We are writing a new version of our [`Bezier`](https://github.com/InstitutMaupertuis/bezier) project that includes a `rviz_visual_tools` object to publish many useful things.

I get this strange error when running our code:

```bash
bezier_application: /usr/include/eigen3/Eigen/src/SVD/JacobiSVD.h:632: const MatrixVType& Eigen::JacobiSVD<MatrixType, QRPreconditioner>::matrixV() const [with _MatrixType = Eigen::Matrix<double, 3, 3>; int QRPreconditioner = 2; Eigen::JacobiSVD<MatrixType, QRPreconditioner>::MatrixVType = Eigen::Matrix<double, 3, 3>; typename _MatrixType::Scalar = double]: Assertion `computeV() && "This JacobiSVD decomposition didn't compute V. Did you ask for it?"' failed.
```

- [rviz_visual_tools pointer member in abstract class](https://github.com/InstitutMaupertuis/bezier/blob/indigo-devel-experimental/bezier_library/include/bezier_library/bezier_library.hpp#L103)
- [pointer init in abstract class constructor](https://github.com/InstitutMaupertuis/bezier/blob/indigo-devel-experimental/bezier_library/src/bezier_library.cpp#L8)
- [concrete inherited class constructor calls base class constructor](https://github.com/InstitutMaupertuis/bezier/blob/indigo-devel-experimental/bezier_library/src/bezier_grinding_surfacing.cpp#L10)
- [rviz_visual_tools usage](https://github.com/InstitutMaupertuis/bezier/blob/indigo-devel-experimental/bezier_library/src/bezier_grinding_surfacing.cpp#L75) - if un-commeted; crash above.

Pointers / ideas are welcome, I'm not sure if the problem lays in `rviz_visual_tools` or our package.
I'll try to reduce this to a minimal example.

davetcoleman, 2016-06-28:
Hmmm not sure why you'd get a Jacobian error - I'm not really using those. I reviewed it all just now and nothing jumped out. 

Just fyi I did a big [cleanup](https://github.com/davetcoleman/rviz_visual_tools/commit/d8e2ffc98464c662c259969291ae52315400b9c5) yesterday - I'm jumping on the chance to break a little bit of API before I release Kinetic. Nothing major in there except I unified the scaling of enum sizes (i.e. SMALL, MEDIUM, LARGE, etc)

If you get a min example I can take a look again

KevinBollore, 2016-06-29:
Hi all!
It works on my machine. I've made some tests with @VictorLamoine code, no JacobiSVD error appends...

VictorLamoine, 2016-06-29:
Such a dumb error!!! There's no problem on `rviz_visual_tools`... I'm trying to copy the first element of an empty vector!

https://github.com/InstitutMaupertuis/bezier/blob/indigo-devel-experimental/bezier_application/src/bezier_application.cpp#L119

I thought it came from `rviz_visual_tools` because of the specific logic I had to use in order to test.
Sorry for the waste of time :flushed:

closed 2016-06-29 by VictorLamoine