PickNikRobotics/rviz_visual_tools #49: Default constructor is not nodelet friendly

shaun-edwards, 2016-12-31:
The `rviz_visual_tools` constructor does not allow the node handle to be passed into the class, rather it initializes the node handle `nh_("~")` [here](https://github.com/davetcoleman/rviz_visual_tools/blob/kinetic-devel/src/rviz_visual_tools.cpp#L55).  When using nodelets, this results in topics being published under an anonymous namespace (which changes every time the nodelet is launched).

I believe the appropriate fix is to allow the node handle to be passed in on construction.  When using nodelets, the user should pass in `getNodeHandle()`.

shaun-edwards, 2016-12-31:
I'm happy to submit a PR to address this, if indeed I am correct and it is an issue.

Would you accept a PR against `indigo`?

davetcoleman, 2016-12-31:
I'd accept it to indigo. Make sure the new constructor has a default value for ``nh`` if the user does not specify. We'll then cherry-pick it to J/K

davetcoleman, 2017-12-25:
Any progress on this or should this issue be closed?

shaun-edwards, 2017-12-26:
I plan to get back to this in the new year, but no longer on indigo.  @geoffreychiou...do you want to have a look at this?

geoffreychiou, 2017-12-27:
@shaun-edwards On kinetic?

shaun-edwards, 2017-12-27:
Kinetic...it looks like this is the latest [LTS](http://wiki.ros.org/Distributions)

VictorLamoine, 2019-02-07:
@davetcoleman you forgot to close this issue

closed 2019-02-23 by davetcoleman