PickNikRobotics/rviz_visual_tools #50: Enhancements in enums

VictorLamoine, 2017-02-03:
Let's say I have a service to publish a marker with a given color and size.

My service would look like:
```
Pose where_to_draw
uint32 color
uint32 size
---
```

The service implementation needs to convert the unsigned int color to a rviz visual tool color; this can be done with
```bash
static_cast<rviz_visual_tools::colors>(req.color)
```

But this results in undefined behavior if `req.color` is not represented in the enum.
A work-around would be to define something like `KEEP_ME_AT_THE_END_ENUM_SIZE` to be able to determine if `color` is in the allowed range. This only works for contingous enums starting from 0.

We would like to avoid a big switch case for each color.

Did you face this issue? How did you solve it? What do you think?

davetcoleman, 2017-03-11:
I'm not sure if this answers your question, but my ``rviz_visual_tools::color`` type is a quick way to choose a color, but it is limiting to what colors you can choose from. For this reason, most of the publish functions also accept ``const std_msgs::ColorRGBA &color``.

But if you must use an ``int``, I have a function for that helpful for libraries such as OMPL:

```
 /** \brief Used by interfaces that do not directly depend on Rviz Visual Tools, such as OMPL */
colors intToRvizColor(std::size_t color);
```

VictorLamoine, 2017-06-01:
Well it's been a long time but your answer didn't fall on deaf ears!
https://gitlab.com/InstitutMaupertuis/ros_additive_manufacturing/merge_requests/28/diffs

Thank you for your help

closed 2017-06-01 by VictorLamoine