PickNikRobotics/rviz_visual_tools #58: Warnings in test compilation

VictorLamoine, 2017-08-03:
Lots of them:
```bash
/home/victor/code/catkin_workspace/src/rviz_visual_tools/tests/rvt_test.cpp: In member function ‘virtual void RVTTest_test_rpy_conversions_Test::TestBody()’:
/home/victor/code/catkin_workspace/src/rviz_visual_tools/tests/rvt_test.cpp:128:82: warning: ‘static Eigen::Affine3d rviz_visual_tools::RvizVisualTools::convertFromXYZRPY(std::vector<double>)’ is deprecated [-Wdeprecated-declarations]
   Eigen::Affine3d expected_affine2 = base.visual_tools_->convertFromXYZRPY(xyzrpy);
                                                                                  ^
```

davetcoleman, 2017-08-10:
Yes I recently made a change that deprecated the ambiguous version of convertFromXYZRPY() and it now requires you say what kind of Euler angles you are using:

> convertFromXYZRPY(x, y, z, roll, pitch, yaw, rviz_visual_tools::XYZ);

See https://github.com/davetcoleman/rviz_visual_tools/blob/kinetic-devel/include/rviz_visual_tools/rviz_visual_tools.h#L949

If you could make that change that would be awesome - my current projects are not using rviz_visual_tools.

VictorLamoine, 2017-08-11:
After the holidays then :)

PR #60 (2017-09-04): Fix deprecated calls to convertFromXYZRPY
Fixes #58
closed 2017-09-04 by davetcoleman