PickNikRobotics/rviz_visual_tools #71: Missing vtable on OSX Build

keshavchintamani, 2018-02-21:
Hi @davetcoleman ,

I'm trying to compile MoveIt on my OSX machine and running across an error while building rviz_visual_tools, which was required by moveit. It seems like the issue has to do with Qt but I cant put my finger on it. Does this have to do with comment in the package.xml

```
<!-- Something changed in ROS Lunar / Ubuntu Zesty that requires extra QT5 dependency-->
  <!-- I suspect this depend is more than needed, but was the only dependency available -->
  <!-- from the rosdep list that worked: https://github.com/ros/rosdistro/blob/master/rosdep/base.yaml -->
  <depend>libqt5x11extras5-dev</depend>
```

My settings:
OSX Sierra 10.12.1
Homebrew 1.5.4
cmake version 3.10.2
Python 3.6.4
Ros Kinetic 

I have rosdep'ed my source and all other moveit ros packages build fine. Other qt dependent packages, e.g. ROS-Desktop-Full were successfully built.
```
Scanning dependencies of target rviz_visual_tools
[100%] Building CXX object rviz_visual_tools/CMakeFiles/rviz_visual_tools.dir/src/rviz_visual_tools.cpp.o
[100%] Building CXX object rviz_visual_tools/CMakeFiles/rviz_visual_tools.dir/src/tf_visual_tools.cpp.o
[100%] Building CXX object rviz_visual_tools/CMakeFiles/rviz_visual_tools.dir/rviz_visual_tools_autogen/mocs_compilation.cpp.o
[100%] Linking CXX shared library /Users/kchintamani/sandbox/moveit_ws/devel/lib/librviz_visual_tools_gui.dylib
Undefined symbols for architecture x86_64:
  "vtable for rviz_visual_tools::RvizVisualToolsGui", referenced from:
      rviz_visual_tools::RvizVisualToolsGui::RvizVisualToolsGui(QWidget*) in rviz_visual_tools_gui.cpp.o
  NOTE: a missing vtable usually means the first non-inline virtual member function has no definition.
  "vtable for rviz_visual_tools::KeyTool", referenced from:
      rviz_visual_tools::KeyTool::KeyTool() in key_tool.cpp.o
      rviz_visual_tools::KeyTool::~KeyTool() in key_tool.cpp.o
  NOTE: a missing vtable usually means the first non-inline virtual member function has no definition.
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [/Users/kchintamani/sandbox/moveit_ws/devel/lib/librviz_visual_tools_gui.dylib] Error 1
make[1]: *** [rviz_visual_tools/CMakeFiles/rviz_visual_tools_gui.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/Users/kchintamani/sandbox/moveit_ws/src/rviz_visual_tools/src/rviz_visual_tools.cpp:56:68: warning:
      suggest braces around initialization of subobject [-Wmissing-braces]
  ...= { RED,        GREEN,  BLUE,   GREY,   DARK_GREY,
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1 warning generated.
[100%] Linking CXX shared library devel/lib/librviz_visual_tools.dylib
[100%] Built target rviz_visual_tools
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
```
Any suggestions are welcome

davetcoleman, 2018-02-26:
I am not sure what the fix would be for this but welcome PRs

keshavchintamani, 2018-03-02:
@davetcoleman I thought it might have something to do with the compiler not using c++11 so I added this in:
include(CheckCXXCompilerFlag)
check_cxx_compiler_flag(-std=c++11 HAS_STD_CPP11_FLAG)
if(HAS_STD_CPP11_FLAG)
  add_compile_options(-std=c++11)
endif()

but that doesnt seem to fix it. I've read elsewhere that this could have something to do with the moc generation via qmake:

https://stackoverflow.com/questions/23595961/qt-a-missing-vtable-usually-means-the-first-non-inline-virtual-member-function

 but honestly I'm far away from fixing anything this deep in cmake.

davetcoleman, 2018-03-08:
Not sure, I only target Ubuntu and don't have the resources to also support OSX but again, contributions are welcome ;)

v4hn, 2018-04-05:
I just experienced this on Lunar-Linux, so it is not specific to Mac.
I have no idea what Ubuntu did to work around this bug. See #74 for the fix.

keshavchintamani, 2018-04-06:
@horagong thanks for the info! I knew the //QObject would fix it. Did you successfully compile moveit on your osx machine using AppleClang? I've been struggling with it for a while now. 

I get past the rviz_visual_tools workaround fix, but I hit this issue:

[ 69%] Building CXX object moveit/moveit_ros/perception/point_containment_filter/CMakeFiles/moveit_point_containment_filter.dir/src/shape_mask.cpp.o
clang: error: no such file or directory: 'NOTFOUND'

This led me to discover OpenMP is not available in AppleClang. I then compiled the Ros libs with LLVM 5.0.1 from HomeBrew which supports OpenMP. The issue agains appears with RVIZ_VISUAL_TOOLS. I keep running into this error:

CMake Error in rviz_visual_tools/CMakeLists.txt:
  No known features for CXX compiler
  "Clang"
  version 5.0.1.

-- Build files have been written to: /Users/kchintamani/sandbox/roskinetic_ws_llvm/moveit_ws/build
Invoking "cmake" failed

keshavchintamani, 2018-04-06:
### Description
Copy pasting issue to here: 

I have successfully compiled ros-desktop-full from source on my mac using LLVM/Clang 5.0.1. I used the homebrew LLVM because some packages in MoveIt required OpenMP, which is not provided by AppleClang.

I've hit a roadblock with this issue with  rviz_visual_tools (also see https://github.com/PickNikRobotics/rviz_visual_tools/issues/71 ) 

```
CMake Error in rviz_visual_tools/CMakeLists.txt:
  No known features for CXX compiler

  "Clang"

  version 5.0.1.
```

I've looked all around for a solution but I cant seem to figure it out. The only related issue I could find was: 

https://cmake.org/pipermail/cmake/2016-December/064722.html

In my case: 
build/rviz_visual_tools/CMakeFiles/ is empty

I tried to pass in these cmake args
-DCMAKE_POLICY_DEFAULT_CMP0025=NEW -DCMAKE_FIND_FRAMEWORK=LAST 

but no luck.

I've tried both catkin_make and catkin tools to build. 

### Your environment
* ROS Distro: Kinetic
* OS Version: macOS Sierra V.10.12.1
* Source build 
* CMake - 3.10
* LLVM/Clang 5.0.1 

* If source, which git commit or tag?
For rviz_visual_tools : origin/kinetic-devel
for moveit : origin/kinetic-devel

### Steps to reproduce
Build ros-kinetic-desktop-full on OSX with latest Homebrew LLVM/Clang 5.0.1 (with toolchain)

export CC=$(brew --prefix llvm)/bin/clang-5.0
export CXX=$(brew --prefix llvm)/bin/clang++
export CPPFLAGS="-I/usr/local/opt/llvm/include -I/usr/local/opt/llvm/include/c++/v1/"
export LDFLAGS="-L/usr/local/opt/llvm/lib -Wl,-rpath,/usr/local/opt/llvm/lib"

Build ros_control 
Build move_it

### Actual behaviour
CMake Error in rviz_visual_tools/CMakeLists.txt:
  No known features for CXX compiler

  "Clang"

  version 5.0.1.

horagong, 2018-04-08:
@keshavchintamani 
I am running moveit successively on mac.
The changes that I made for compiling are in https://github.com/ros-planning/moveit/issues/818

For compiling rviz_visual_tools, you should take [#74](https://github.com/PickNikRobotics/rviz_visual_tools/pull/74) fix. Just commenting out QObject makes some features do not work.

keshavchintamani, 2018-04-10:
@horagong  Many thanks for the support! I have now successfully compiled moveit! Besides commenting out QObject in rviz_visual_tools, I got past the issue of LibOMP like this (using AppleClang): 

> brew install libomp
> catkin_make --cmake-args "-DOpenMP_CXX_FLAGS=-Xpreprocessor" "-DOpenMP_CXX_FLAGS=-fopenmp" "-DOpenMP_CXX_FLAGS=-I/usr/local/opt/libomp/include" "-DOpenMP_CXX_LIB_NAMES=omp" "-DOpenMP_omp_LIBRARY=/usr/local/opt/libomp/lib/libomp.dylib"

closed 2018-04-10 by keshavchintamani