PickNikRobotics/rviz_visual_tools #75: Compiling Errors

dhiegomaga, 2018-04-09:
On my first trial to use the library, for ROS Kinetic on Ubuntu 16.04. 

Installed using apt - ok
`sudo apt-get install ros-kinetic-rviz-visual-tools`

Ran Rviz and the demo - ok
```
roslaunch rviz_visual_tools demo_rviz.launch
roslaunch rviz_visual_tools demo.launch
```

Then I tried the cpp code, and got error... 

Included headers
`#include <rviz_visual_tools/rviz_visual_tools.h>`

Added code to main():
```
rviz_visual_tools::RvizVisualToolsPtr visual_tools_;
visual_tools_.reset(new rviz_visual_tools::RvizVisualTools("base_frame","/rviz_visual_markers"));
```

Got many compile errors: 

[image]
[image]

mcevoyandy, 2018-04-09:
Looks like you'r not compiling with c++11? Did you make the appropriate changes to your cmake and package files in order to use the library?

dhiegomaga, 2018-04-09:
@mcevoyandy No I didn't change anything inside the CMakeLists.txt (is that what you meant?). The tutorial didn't specify anything... What should I change exactly?

mcevoyandy, 2018-04-09:
When you say "I tried the cpp code" I assume you have your own package that you're trying to use this in?

If so, add `rviz_visual_tools` to `find_package` in your `CMakeLists.txt` file and add `rviz_visual_tools` as a depend in your `package.xml` file.

http://wiki.ros.org/ROS/Tutorials

mcevoyandy, 2018-04-10:
Yes, I forgot to mention the C++11, but that one was explicit in your compile errors. If you think the readme could benefit from additional explanation we'd love to review a pull request.

closed 2018-04-10 by mcevoyandy

davetcoleman, 2018-04-10:
@dhiegomaga please contribute to improving the README.md!