PickNikRobotics/rviz_visual_tools #77: No member named trigger in INDIGO?

CesMak, 2018-04-19:
Hey there,

I have kinetic code that i need to be able to run on indigo.

My code runs completely fine on kinetic but on indigo I got this  error:

error: ‘class rviz_visual_tools::RvizVisualTools’ has no member named ‘trigger’
   visual_tools_->trigger();

Code Snipped:
` rviz_visual_tools::RvizVisualToolsPtr visual_tools_; `

CesMak, 2018-04-19:
I installed it via apt-get (as I said i just installed rviz_visual_tools for indigo)

mcevoyandy, 2018-04-19:
Sorry, I just saw that you have kinetic code you need to run on indigo and wanted to clarify.

You will need to install the kinetic-devel branch from source. There are some features that in the kinetic-devel branch that are not available in indigo-devel.

mcevoyandy, 2018-04-24:
Clone the kinetic devel branch into your workspace and compile it with all of your other packages.

davetcoleman, 2018-04-24:
We make breaking changes between ROS versions to improve the overall feature set. This means you cannot use Indigo debian and expect it to work like Kinetic - that ruins the whole point. However like Andy said building from source is a great work around. There are likely other work arounds but we cannot support that ourselves, that's up to you to figure out.

CesMak, 2018-04-25:
All right thanks :) -> I just came around not using rviz_visual_tools it also works for me now :)