PickNikRobotics/rviz_visual_tools #79: qt5_use_modules missing

machinekoder, 2018-06-08:
I'm getting linker errors with Qt 5.10. adding `qt5_use_modules(imarker_simple_demo Widgets)` in the CMakeLists.txt fixes the issue.

```
# Demo executable
add_executable(${PROJECT_NAME}_demo src/${PROJECT_NAME}_demo.cpp)
target_link_libraries(${PROJECT_NAME}_demo
  ${catkin_LIBRARIES} ${PROJECT_NAME}
  )
qt5_use_modules(${PROJECT_NAME}_demo Widgets)

# Demo executable
add_executable(imarker_simple_demo src/imarker_simple_demo.cpp)
target_link_libraries(imarker_simple_demo
  ${catkin_LIBRARIES} ${PROJECT_NAME} ${PROJECT_NAME}_imarker_simple
)
qt5_use_modules(imarker_simple_demo Widgets)
```

machinekoder, 2018-06-08:
`qt_version_tag not found` is the error

VictorLamoine, 2019-02-07:
@machinekoder this issue should be solved, https://github.com/PickNikRobotics/rviz_visual_tools/blob/melodic-devel/CMakeLists.txt#L45

Can you check again and close if it is the case?

machinekoder, 2019-02-07:
Yes, thanks.

closed 2019-02-07 by machinekoder