PickNikRobotics/rviz_visual_tools #81: multiple definition of rviz_visual_tools::KeyTool::qt_static_metacall

HankerSia, 2018-06-19:
Hello, I finished the rosdep install command, and when i tried to catkin_make the workspace, i get follow errors:
 [ 97%] Linking CXX shared library /home/robot/catkin_ws/devel/lib/librviz_visual_tools_gui.so
CMakeFiles/rviz_visual_tools_gui.dir/rviz_visual_tools_gui_automoc.cpp.o: In function `rviz_visual_tools::KeyTool::qt_static_metacall(QObject*, QMetaObject::Call, int, void**)':
rviz_visual_tools_gui_automoc.cpp:(.text+0x0): multiple definition of `rviz_visual_tools::KeyTool::qt_static_metacall(QObject*, QMetaObject::Call, int, void**)'
CMakeFiles/rviz_visual_tools_gui.dir/src/moc_key_tool.cpp.o:moc_key_tool.cpp:(.text+0x0): first defined here
CMakeFiles/rviz_visual_tools_gui.dir/rviz_visual_tools_gui_automoc.cpp.o:(.data.rel.ro+0x20): multiple definition of `rviz_visual_tools::KeyTool::staticMetaObject'
CMakeFiles/rviz_visual_tools_gui.dir/src/moc_key_tool.cpp.o:(.data.rel.ro+0x20): first defined here
CMakeFiles/rviz_visual_tools_gui.dir/rviz_visual_tools_gui_automoc.cpp.o: In function `rviz_visual_tools::KeyTool::metaObject() const':
rviz_visual_tools_gui_automoc.cpp:(.text+0x16): multiple definition of `rviz_visual_tools::KeyTool::metaObject() const'
CMakeFiles/rviz_visual_tools_gui.dir/src/moc_key_tool.cpp.o:moc_key_tool.cpp:(.text+0x16): first defined here
CMakeFiles/rviz_visual_tools_gui.dir/rviz_visual_tools_gui_automoc.cpp.o: In function `rviz_visual_tools::KeyTool::qt_metacast(char const*)':
rviz_visual_tools_gui_automoc.cpp:(.text+0x5e): multiple definition of `rviz_visual_tools::KeyTool::qt_metacast(char const*)'
CMakeFiles/rviz_visual_tools_gui.dir/src/moc_key_tool.cpp.o:moc_key_tool.cpp:(.text+0x5e): first defined here
CMakeFiles/rviz_visual_tools_gui.dir/rviz_visual_tools_gui_automoc.cpp.o: In function `rviz_visual_tools::KeyTool::qt_metacall(QMetaObject::Call, int, void**)':
rviz_visual_tools_gui_automoc.cpp:(.text+0xae): multiple definition of `rviz_visual_tools::KeyTool::qt_metacall(QMetaObject::Call, int, void**)'
CMakeFiles/rviz_visual_tools_gui.dir/src/moc_key_tool.cpp.o:moc_key_tool.cpp:(.text+0xae): first defined here
CMakeFiles/rviz_visual_tools_gui.dir/rviz_visual_tools_gui_automoc.cpp.o: In function `rviz_visual_tools::RvizVisualToolsGui::qt_static_metacall(QObject*, QMetaObject::Call, int, void**)':
rviz_visual_tools_gui_automoc.cpp:(.text+0xee): multiple definition of `rviz_visual_tools::RvizVisualToolsGui::qt_static_metacall(QObject*, QMetaObject::Call, int, void**)'
CMakeFiles/rviz_visual_tools_gui.dir/src/moc_rviz_visual_tools_gui.cpp.o:moc_rviz_visual_tools_gui.cpp:(.text+0x0): first defined here
CMakeFiles/rviz_visual_tools_gui.dir/rviz_visual_tools_gui_automoc.cpp.o:(.data.rel.ro+0x60): multiple definition of `rviz_visual_tools::RvizVisualToolsGui::staticMetaObject'
CMakeFiles/rviz_visual_tools_gui.dir/src/moc_rviz_visual_tools_gui.cpp.o:(.data.rel.ro+0x20): first defined here
CMakeFiles/rviz_visual_tools_gui.dir/rviz_visual_tools_gui_automoc.cpp.o: In function `rviz_visual_tools::RvizVisualToolsGui::metaObject() const':
rviz_visual_tools_gui_automoc.cpp:(.text+0x16c): multiple definition of `rviz_visual_tools::RvizVisualToolsGui::metaObject() const'
CMakeFiles/rviz_visual_tools_gui.dir/src/moc_rviz_visual_tools_gui.cpp.o:moc_rviz_visual_tools_gui.cpp:(.text+0x7e): first defined here
CMakeFiles/rviz_visual_tools_gui.dir/rviz_visual_tools_gui_automoc.cpp.o: In function `rviz_visual_tools::RvizVisualToolsGui::qt_metacast(char const*)':
rviz_visual_tools_gui_automoc.cpp:(.text+0x1b4): multiple definition of `rviz_visual_tools::RvizVisualToolsGui::qt_metacast(char const*)'
CMakeFiles/rviz_visual_tools_gui.dir/src/moc_rviz_visual_tools_gui.cpp.o:moc_rviz_visual_tools_gui.cpp:(.text+0xc6): first defined here
CMakeFiles/rviz_visual_tools_gui.dir/rviz_visual_tools_gui_automoc.cpp.o: In function `rviz_visual_tools::RvizVisualToolsGui::qt_metacall(QMetaObject::Call, int, void**)':
rviz_visual_tools_gui_automoc.cpp:(.text+0x204): multiple definition of `rviz_visual_tools::RvizVisualToolsGui::qt_metacall(QMetaObject::Call, int, void**)'
CMakeFiles/rviz_visual_tools_gui.dir/src/moc_rviz_visual_tools_gui.cpp.o:moc_rviz_visual_tools_gui.cpp:(.text+0x116): first defined here
collect2: error: ld returned 1 exit status
rviz_visual_tools/CMakeFiles/rviz_visual_tools_gui.dir/build.make:288: recipe for target '/home/robot/catkin_ws/devel/lib/librviz_visual_tools_gui.so' failed
make[2]: *** [/home/robot/catkin_ws/devel/lib/librviz_visual_tools_gui.so] Error 1
CMakeFiles/Makefile2:10283: recipe for target 'rviz_visual_tools/CMakeFiles/rviz_visual_tools_gui.dir/all' failed
make[1]: *** [rviz_visual_tools/CMakeFiles/rviz_visual_tools_gui.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 97%] Linking CXX shared library /home/robot/catkin_ws/devel/lib/librviz_visual_tools.so
[ 97%] Built target rviz_visual_tools
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j12 -l12" failed

And i have tried many times, but i did not make it. I use ubuntu 16.04 and ros kinetic.
Looking forward to your advice!

davetcoleman, 2018-06-27:
Yea for Kinetic and Melodic i think we can safely remove QT4

simonschmeisser, 2018-07-03:
note that this issue has been fixed for kinetic in #76 but needs to be forward ported to melodic