PickNikRobotics/rviz_visual_tools #97: Could not find a panel name "RvizVisualToolsGui" in rviz

WwYyFan, 2018-09-30:
I'm using ur robot,and I can't find RvizVisualToolsGui in rviz when I run the following command
`
roslaunch ur3_moveit_config demo.launch
`

VictorLamoine, 2019-02-07:
- Is `rviz_visual_tools` installed on your machine? What version?
  - `dpkg -l | grep rviz-visual-tools`

- If not, is it included in your catkin workspace? What version?
  - `cd catkin/src/rviz_visual_tools && git rev-parse HEAD`

Launch RViz and go to panels > add, you should be able to add the panel:

[image]

VictorLamoine, 2019-08-20:
@davetcoleman I think you can close this issue

davetcoleman, 2019-08-20:
@VictorLamoine you've been a very dedicated contributor to this project, thank you! I've just added you as a Contributor with write access, so you can close these yourself in the future.

Please use your best judgement but feel free to be a full maintainer of this repo, and even add your name to the README next to mine

closed 2019-08-20 by davetcoleman