PickNikRobotics/rviz_visual_tools #118: how to to display plane ax + bx + cy + d = 0

aidenyao, 2019-04-21:


matthaeusheer, 2019-04-29:
Support this question, need to solve the exact same problem.  
I have model coefficients (```pcl::ModelCoefficients::Ptr coefficients(new pcl::ModelCoefficients);```) which I would like to use to publish planes using some sort of translation and rotation.

AndyZe, 2019-05-01:
I started a [branch](https://github.com/PickNikRobotics/rviz_visual_tools/tree/display_plane) for this feature but rotation is not correct yet. [This question](https://answers.ros.org/question/266391/quaternion-to-rotate-rigid-frame/) on answers.ros.org has a solution for the rotation. If you like, you could implement that solution around line 981 of rviz_visual_tools.cpp

AndyZe, 2019-05-03:
@aidenyao @matthaeusheer  Can you please test out PR #119 and let me know if it works for you.

matthaeusheer, 2019-05-03:
Hey, thanks for the work. I will checkout it out tomorrow!

AndyZe, 2019-05-22:
#119

closed 2019-05-22 by AndyZe