PickNikRobotics/rviz_visual_tools #145: publishLine causes warning "Uninitialized quaternion, assuming identity"

CodeFinder2, 2020-02-17:
Hi,

when I am doing:
```cpp
    rviz_visual_tools::RvizVisualToolsPtr visual_tools;
    visual_tools.reset(new rviz_visual_tools::RvizVisualTools(frame_id, rviz_topic_name));
    visual_tools->loadMarkerPub(false, true);
    visual_tools->deleteAllMarkers();
    visual_tools->enableBatchPublishing();

    Eigen::Isometry3d pose1 = Eigen::Isometry3d::Identity();
    Eigen::Isometry3d pose2 = Eigen::Isometry3d::Identity();
    pose1.translation().x() = 1.0;
    pose1.translation().y() = 1.0;
    pose2.translation().x() = 5.0;
    pose2.translation().y() = 5.0;
    visual_tools->publishLine(pose1, pose2, rviz_visual_tools::RAND, rviz_visual_tools::LARGE);
    visual_tools->trigger();
```
Rviz shows:
> Line/1
> Uninitialized quaternion, assuming identity. scale.y and scale.z of LINE_LIST or LINE_STRIP are ignored.

However, the line is shown correctly. This also happens using the `geometry_msgs::Point` variant of `publishLine()`.

JafarAbdi, 2020-04-15:
Hi @CodeFinder2, Are you building the package from source or installing it from apt .? this got fixed for the master in this PR https://github.com/PickNikRobotics/rviz_visual_tools/pull/129

CodeFinder2, 2020-04-15:
Ah yes I installed via apt (Ubuntu 18.04), maybe the packages are still outdated.

Thanks for the hint! Will close this issue.

closed 2020-04-15 by CodeFinder2