PickNikRobotics/rviz_visual_tools #146: trigger() blocks forever - bug?

CodeFinder2, 2020-02-17:
Hi,

I am facing the problem that
```cpp
    ROS_ERROR("FOO");
    visual_tools->trigger();
    ROS_ERROR("BAR");
```
does not return. Even when Rviz connects, nothing will happen. I don't know if this is related to my issue but note that neither `visual_tools->loadMarkerPub(false, true);` nor `visual_tools->loadMarkerPub(true, true);` is working. (I am requiring latched topics.)

Once I comment this block:
```cpp
  // Check if connected to a subscriber
  if (!pub_rviz_markers_waited_ && !pub_rviz_markers_connected_)
  {
    ROS_DEBUG_STREAM_NAMED(LOGNAME, "Waiting for subscribers before publishing markers...");
    waitForSubscriber(pub_rviz_markers_);

    // Only wait for the publisher once, after that just ignore the lack of connection
    pub_rviz_markers_waited_ = true;
  }
```
in the [implementation of `RvizVisualTools::publishMarkers()`](https://github.com/PickNikRobotics/rviz_visual_tools/blob/master/src/rviz_visual_tools.cpp#L891) (invoked by `trigger()`), it is working fine.

Is this a bug? IMHO, there should be the possibility to skip the `waitForSubscriber()`.

CodeFinder2, 2020-02-17:
Okay, nevermind: it is related to the `ros::spinOnce()` in `RvizVisualTools::waitForSubscriber()`. I have previously locked a mutex and if `spinOnce()` invoked a callback (locking the mutex again), it deadlocks.

So, would it be possible to skip `waitForSubscriber()`? Probably adding an additonal parameter?

mfocchi, 2020-11-19:
Me and all my research group have the exact same issue. Is it going to be fixed? In which version?

PR #193 (2022-10-05): Move pub_rviz_markers_ to it's own callback queue
waitForSubscriber used to call ros::spinOnce() which could trigger
callbacks from other topics in other threads. In case those have a lock
to only get called once, this would look the visual_tools thread.

Closes: #146
commit 74b92d85 (2022-10-05): Move pub_rviz_markers_ to it's own callback queue (#193)

waitForSubscriber used to call ros::spinOnce() which could trigger
callbacks from other topics in other threads. In case those have a lock
to only get called once, this would look the visual_tools thread.

Closes: #146
closed 2022-10-05 by JafarAbdi