PickNikRobotics/rviz_visual_tools #148: `generateEmptyPose` suddenly removed

Dale-Koenig, 2020-03-18:
`static void generateEmptyPose(geometry_msgs::Pose& pose);` was removed in the latest master without deprecating it.  This caused a few problems (it was used, for example, in moveit grasps).  However, any active projects using the master branch have likely already updated, or are not using the function to begin with, so I am not sure it is worth putting back in.

That said, the sudden removal causes some complications with backwards compatibility, because the replacement `static geometry_msgs::Pose getIdentityPose();` did not exist in the preivous release.  Therefore, repositories using `master` will not compile with `melodic-devel` and vice versa. Perhaps we could add in `static geometry_msgs::Pose getIdentityPose();` to `melodic-devel`, and add a deprecated attribute to `generateEmptyPose`?

Dale-Koenig, 2020-05-14:
@davetcoleman @mlautman There are several people complaining about problems related to this in the Moveit Grasps repo.

davetcoleman, 2020-06-02:
Hi @Dale-Koenig ,

I'm not really involved with development anymore :-(
But I see [this commit](https://github.com/PickNikRobotics/rviz_visual_tools/commit/c67d05cbe3d8340d02a8cbd82ec336caaa4f0adb) you are referring to. Can you create a PR to backport that to melodic-devel and the other requested changes?

mlautman, 2020-06-02:
@Dale-Koenig I believe that the master branch of moveit  grasps has been patched. Re-open if I am wrong about this

Dale-Koenig, 2020-06-02:
> @Dale-Koenig I believe that the master branch of moveit grasps has been patched. Re-open if I am wrong about this

The problem is when using melodic branches.  With both on master it is fine, but the melodic branches are incompatible

davetcoleman, 2020-06-04:
Sorry for the confusion!

>  Can you create a PR to backport that to melodic-devel and the other requested changes?

PR #156 (2020-06-09): Backport get_identity_pose
Close #148