PickNikRobotics/rviz_visual_tools #178: Creating foxy branch

MoriKen254, 2021-01-22:
Could you please create the foxy branch, because I would like to send a pull request for foxy?

I already confirmed this repo worked on Foxy after a bit correction.

## Progress
### Build error
First, I tried to build the eloquent-devel source code, but build failed. The error message is as follows.

```bash
/rviz_visual_tools/src/tf_visual_tools.cpp:63:98: error: no matching function for call to ‘create_timer(std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface, gnu_cxx::_S_atomic>::element_type, std::shared_ptr<rclcpp::node_interfaces::NodeTimersInterface, gnu_cxx::_S_atomic>::element_type, rclcpp::Clock::SharedPtr, rclcpp::Duration&, std::_Bind_helper<false, void (rviz_visual_tools::TFVisualTools::)(), rviz_visual_tools::TFVisualTools>::type)’
   63 |                                             std::bind(&TFVisualTools::publishAllTransforms, this));
```

### Comparison of API b/w foxy and eloquent
- http://docs.ros2.org/foxy/api/rclcpp/create__timer_8hpp.html
- https://docs.ros2.org/eloquent/api/rclcpp/create__timer_8hpp.html

Specification of arguments in `rclcpp::create_timer` were updated from offering a `raw pointer` to a `shared_poniter`.

### Correction
Then I edited the arguments of the `create_timer` as follows:

https://github.com/MoriKen254/rviz_visual_tools/commit/80135301b44bd0a845f0edc58463418b9452c55b

### Test
Then  I run test program and confirmed the appropriate functionality.

[image]

As soon as you create the foxy-devel branch, please let me create a pull request on this issue.

JafarAbdi, 2021-01-22:
@MoriKen254 Here you go https://github.com/PickNikRobotics/rviz_visual_tools/tree/foxy-devel, please let me know when you open the PR so I review and merge it

Thanks!

davetcoleman, 2021-01-23:
Thanks for creating the branch @JafarAbdi !

closed 2021-01-23 by davetcoleman

MoriKen254, 2021-01-23:
@JafarAbdi Thank you! I open the PR at https://github.com/PickNikRobotics/rviz_visual_tools/pull/179