PickNikRobotics/rviz_visual_tools #210: RViz Visual Tools is failing in buildfarm for Foxy

vatanaksoytezer, 2021-12-01:
rviz-visual-tools have been failing in buildfarm for a long time, see: https://build.ros2.org/job/Fbin_uF64__rviz_visual_tools__ubuntu_focal_amd64__binary/247/console for context. Error seems to be:

```
23:03:59 CMake Error at CMakeLists.txt:39 (find_package):
23:03:59   By not providing "Findrviz_common.cmake" in CMAKE_MODULE_PATH this project
23:03:59   has asked CMake to find a package configuration file provided by
23:03:59   "rviz_common", but CMake did not find one.
23:03:59 
23:03:59   Could not find a package configuration file provided by "rviz_common" with
23:03:59   any of the following names:
23:03:59 
23:03:59     rviz_commonConfig.cmake
23:03:59     rviz_common-config.cmake
23:03:59 
23:03:59   Add the installation prefix of "rviz_common" to CMAKE_PREFIX_PATH or set
23:03:59   "rviz_common_DIR" to a directory containing one of the above files.  If
23:03:59   "rviz_common" provides a separate development package or SDK, be sure it
23:03:59   has been installed.
```

vatanaksoytezer, 2021-12-01:
@tylerjw @henningkayser should I add pre-release and send a fix to this?

JafarAbdi, 2021-12-01:
We need to backport https://github.com/PickNikRobotics/rviz_visual_tools/pull/209 to foxy and cut a new release

vatanaksoytezer, 2021-12-06:
> We need to backport #209 to foxy and cut a new release

Yep, we should cut a release for all distros though, we didn't release rvt for galactic / rolling after #209.

vatanaksoytezer, 2022-03-31:
The builds are passing now.

closed 2022-03-31 by vatanaksoytezer